openmv/openmv-ide #144: "Save open script to openMV cam" menu button greyed out

kdarras, 2022-03-03:
I have a feeling this happens since the latest IDE was released.
The button is regularly grey and unselectable with several, but not all of my H7+ boards, even though they are connected to the IDE, whether there is a main.py file already on the SD card or not.
Running Ubuntu 20.04

kwagyeman, 2022-03-03:
This is not related to the IDE release. It's just the IDE can't find your SD card.

kwagyeman, 2022-03-03:
Please fix the file system on the SD card to be FAT32

kdarras, 2022-03-04:
All my SD cards are in FAT32:
[image]
I only experienced this problem recently, I think it did not happen with previous IDE versions.

kdarras, 2022-03-06:
This issue was observed on several Windows computers as well.
The file SD card's file format was not the problem. Instead, the hidden .opemv_disk file was missing (maybe deleted by users mistakenly). Copying it to the SD card instantly solved the issue. Is there a way to have the IDE to re-generate that file?

kwagyeman, 2022-03-06:
It's generated on a fresh disk of the OpenMV Cam @iabdalkader .

kwagyeman, 2022-03-06:
This is a firmware issue. The IDE looks for this file to verify a harddrive is attached,