openmv/openmv-ide #145: Nano RP2040 Connect Micropython MLC example OSError: [Errno 2] ENOENT

jomoengineer, 2022-03-03:


kwagyeman, 2022-03-03:
Ibrahim, please rename the example script in our repo. I will then update the examples with the new firmware release.

iabdalkader, 2022-03-03:
I can rename it, but the issue is it's copying the example not the library from `libraries`.

jomoengineer, 2022-03-03:
@iabdalkader 
Yes, I forgot to add the step for the model but I did find it and run it.

Initially I downloaded the lsm6dsox.py lib from the following MicroPython link since I was not able to find it elsewhere.
Note: They call the basic lsm6dsox example lsm6dsox_basic.py.  The way it is listed in OpenMV IDE both the lib code and the example are the same name. 
[https://github.com/micropython/micropython/tree/master/drivers/lsm6dsox](url)

Perhaps this is why the example started to work.

Also, there is a second pop-up that appears after it asks to download an update to the lsm6dsox.py on the board which asks if the user want to update the file on the PC. 
[image]

 If I do this with the lsm6dsox.py lib code on the board, and then restart the IDE, I do not see the request to download the file to board or the error.

iabdalkader, 2022-03-03:
Renamed the example to 'lsm6dsox_basic.py`

jomoengineer, 2022-03-03:
I renamed the lsm6dsox.py example to lsm6dsox_basic.py under the 'share\qtcreator\examples\Arduino\Nano-RP2040\01-Sensors' OpenMV IDE folder, and then:
 1. Started OpenMV IDE
 2. Connected to the Rp2040
 3. Reflashed the FW on the RP2040 from the IDE including erasing the internal mem
 4. Safe Removed and re inserted the RP2040 USB cable
 5. Copied the lsm6dsox_vibration_monitoring.ucf model file to the RP2040
 6. Reran the lsm6dsox_mlc.py example and had the IDE copy the lsm6dsox driver to the RP2040.
 7. This did result in what looks like the driver code being copied to the board but did result in an error in the IDE:
 
 ```
Traceback (most recent call last):
  File "<stdin>", line 28, in <module>
  File "lsm6dsox.py", line 136, in __init__
  File "lsm6dsox.py", line 189, in load_mlc
OSError: [Errno 2] ENOENT
MPY: soft reboot
MicroPython: v1.17-omv-r22 OpenMV: v4.2.3 HAL: v1.3.0 BOARD: Arduino Nano RP2040 Connect-RP2040
Type "help()" for more information.
>>> 
```
 It stops at the following code from line 129 in lsm6dsox.py file from the board

```
	def read_accel(self):
		mv = memoryview(self.scratch_int)
		f = self.accel_scale
		self.i2c.readfrom_mem_into(self.address, _OUTX_L_XL, mv)
		return (mv[0] / f, mv[1] / f, mv[2] / f)

```

If I click on the lsm6dsox_mlc.py example window in the IDE and run it again, it seems to work just fine.

It does prove that the correct lsm6dsox.py driver does get copied to the board with the lsm6dsox.py example named lsm6dsox_basic.py.

iabdalkader, 2022-03-03:
Thanks for testing! I already renamed it in the other repo, and this should be fixed in the next update. FYI after copying a file to the storage, you need to reset the cam. The IDE should probably mention that.

jomoengineer, 2022-03-03:
Yeah, it just ran and produced the error after I clicked ok to copy the file to the board.  It did not stop the run process after copying the file.

iabdalkader, 2022-03-04:
This one needs to be fixed @kwagyeman. The cam doesn't see new files before a reset, maybe force a reset ?

iabdalkader, 2022-03-04:
You know actually I just realized that this library is already built-in (frozen) in the Nano RP2040 firmware, the IDE just needs to learn that and ignore copying it. You shouldn't need to copy it to the file system at all.

kwagyeman, 2022-03-04:
Yeah, the IDE won't ask unless you name an example the same as a library.

iabdalkader, 2022-03-04:
It's true, it does not ask to copy. I guess this issue is fixed.

closed 2022-03-04 by iabdalkader