openmv/openmv-ide #147: Portenta load firmware fails.

iabdalkader, 2022-03-09:
On Linux, even though the board resets to DFU.

[image]

kwagyeman, 2022-04-24:
This happens because the PID/VIDs are screwed up that the board is returning. For example, I can confirm on the Nicla that it has the PID 045f in application mode which is the bootloader PID.

kwagyeman, 2022-04-24:
I will add the bootloader PID to the list of PIDs that the IDE will accept... however, this isn't really right.

kwagyeman, 2022-04-24:
Wait, no, this is correct.

iabdalkader, 2022-04-25:
> I will add the bootloader PID to the list of PIDs that the IDE will accept...

If you run it again it works. I think it just needs to wait longer or poll to account for different USB hubs/controllers timing.

iabdalkader, 2022-04-25:
Note the Portenta and NiclaV have different bootloader PIDs...So is this issue fixed or not ?

kwagyeman, 2022-04-25:
Yes

iabdalkader, 2022-05-24:
Fixed in 2.9.2

closed 2022-05-24 by iabdalkader