openmv/openmv-ide #149: Tab bar disappears under certain conditions.

iabdalkader, 2022-03-10:
* Run the IDE. By default it opens `helloworld_1.py`
* Make any change to this script **without saving** (for example delete any line)
* Close the script by clicking the **X on the script tab**.
* The IDE shows the dialog that asks to save changes.
* Press the keyboard escape key or click on dialog's Cancel button (do Not click "Do not save").
* The tab bar disappears:
[image]

kwagyeman, 2022-04-24:
This is normal. The Tab bar disappears when no scripts are open. It will reappear when you open a script.

kwagyeman, 2022-04-24:
I don't see a bug here. Just tested it out.

iabdalkader, 2022-04-25:
> The Tab bar disappears when no scripts are open

But the script is still open.

[image]

kwagyeman, 2022-04-25:
I see the issue now. I clicked accept.

kwagyeman, 2022-04-25:
Found the issue, fixed it.

iabdalkader, 2022-04-25:
> I see the issue now. I clicked accept.

That's why I was very specific about the steps :smiling_face_with_tear:

iabdalkader, 2022-05-23:
Related issues: Unchanged tab closes together with unsaved tab, and the bar still disappears under certain conditions.

### Please take the time to read and follow the steps carefully to reproduce these issues:

* Start the IDE (by default it opens `helloworld_1.py`)
* Create 2 new scripts (hit Ctrl+n twice).
* Now you have 3 open scripts: `helloworld_1.py`, `untitled_1.py` and `untitled_2.py`
* Make any change to the **second script** `untitled_1.py` (i.e the one before the last open script).
* Close that script (`untitled_1.py`) by clicking the **X on the script tab**.
* The save script popup dialog shows up, click on the **"Do not save"** button.
#### First bug -> the last two tabs are closed (`untitled_1.py` and `untitled_2.py`).

* Next close the remaining `hellowrld_1.py` script by clicking the **X on the script tab**.
#### Second bug -> script is not closed and tab bar disappears.

kwagyeman, 2022-05-23:
Is this with the new version of the compiled IDE?

kwagyeman, 2022-05-24:
Found the bug. Code was calling itself twice.