openmv/openmv-ide #152: Stopping Script in the middle of printing crashes firmware.

kwagyeman, 2022-04-02:
Run this:

```
# Hello World Example
#
# Welcome to the OpenMV IDE! Click on the green run arrow button below to run the script!

import sensor, image, time

sensor.reset()                      # Reset and initialize the sensor.
sensor.set_pixformat(sensor.RGB565) # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QVGA)   # Set frame size to QVGA (320x240)
sensor.skip_frames(time = 2000)     # Wait for settings take effect.
clock = time.clock()                # Create a clock object to track the FPS.

while(True):
    clock.tick()                    # Update the FPS clock.
    #img = sensor.snapshot()         # Take a picture and return the image.
    print(clock.fps())              # Note: OpenMV Cam runs about half as fast when connected
                                    # to the IDE. The FPS should increase once disconnected.

```

Press stop when it starts printing a lot of stuff. On the H7 Plus you crash when this happens. The IDE never gets the USB packet completion from the camera and locks up.

kwagyeman, 2022-04-24:
The IDE now idles commands to the camera before stopping, stops the camera, then drains the tx buffer, then drains the frame buffer.

This should fix the issue...

kwagyeman, 2022-04-24:
Note, the IDE was always polling the text buffer before... this means this issue would happen if the IDE sent another command except draining the text buffer... or it means the camera firmware is bugged.

iabdalkader, 2022-04-25:
@kwagyeman Want to explain more ? Note I traced the code and it was blocking waiting for the text buffer to drain. Note that right after the IDE sends a STOP, an exception is raised `mp_const_ide_interrupt` and it's printed, that's what causes the block.

kwagyeman, 2022-04-25:
The IDE will issue commands to drain the text buffer right after stop is issued.

However, the IDE previously did this but also had the dump frame and get script running mixed in there too.

So... given all of the previous commands should work I don't know if just only issuing commands to drain the text buffer will fix the issue.

iabdalkader, 2022-04-25:
I just realized that when