openmv/openmv-ide #153: Text editor's font size only has few options for WIndows

kdarras, 2022-05-20:
On a windows computer, the openMV IDE text editor options only offer font sizes starting at 13 and with increments of 2 or 3 (cannot remember exactly). Impossible to input a custom value, the Text editor' s font size remains impractically large.

kwagyeman, 2022-05-20:
Use the zoom feature to zoom in or out.

closed 2022-05-20 by kwagyeman

kdarras, 2022-05-20:
Maybe this should not be closed yet, as the zoom feature, after clicking "apply", did not change anything.

kdarras, 2022-05-21:
Only the text size of the row line numbers change proportionally with the zoom value.
The text editor font size can only increase above 100%, it does not decrease below 100%, whether we use the options of the mouse wheel, same thing.