openmv/openmv-ide #157: OpenMV IDE Customerzation for Laminating machine vision two-offset breakpoint system

Undertaker7533967, 2022-06-24:
1.ROI_A and ROI_B gap X, it could be adjusted by (button +) or (button -) the step resolution is 0.1mm
2.ROIA and ROIB width Y, it could be adjusted by (button +) or (button -) the step resolution is 0.1mm
3.Up and Down location setting Up: Z, Down: W , it could be adjusted by (button +) or (button -)
the step resolution is 0.1mm
4. Star button/Stop button
Outputs condition:
1. No outputs :Both NIKE label into ROI_A and ROI_B at same time
2. No outputs :Neither NIKE label into ROI_A and ROI_B
3. If NIKE label into ROI_A , outputs Pin A immediately.
4. If NIKE label into ROI_B, outputs Pin B immediately.
5. image FOV:15mmx100mm
as attached picture below:
[[image]
](url)

Undertaker7533967, 2022-09-28:
Hallo Kwagyeman:

We have some question is how to implement threshold editor(Lmin, Lmax, Amin, Amax, Bmin, Bmax) from OpenMV IDE to our requirement GUI??
Please provide us some design guide kindly~~
all the best~~

kwagyeman, 2022-09-29:
The code is here: https://github.com/openmv/qt-creator/blob/788fcebb24a193e4ddf9105adb4fef2333cb8e1f/src/plugins/openmv/tools/thresholdeditor.cpp

Undertaker7533967, 2022-09-30:
Dear Kwagyeman:

Thank you for threshold.cpp support ~~now ,we want to build our own GUI include ROI , threshold combine with openMV cam h7 plus~~if we have other question please assist us again~~
all the best~~

kwagyeman, 2023-03-29:
Hi, we have a bayer version of the camera. The driver is already done and the test samples work. We can build them for you. Note that while it works in color the image size is forced to be like VGA and then you have to crop.

As for the unit count, I need a MOQ of 250 units. If that is fine we can build these for you ASAP and have them shipped to you within 3 months.

Undertaker7533967, 2023-03-30:
Hallo Kwagyeman:

We are glade to hear the good news from you, We can buy your one sample and testing on our customer first.
After testing done, we will order some, but still want to know the quotation of 250 units.   

all the best~~

kwagyeman, 2023-03-30:
Please email us. This a public forum. The price would be 30% off the list price online.

kwagyeman, 2023-07-18:
Public comment on this issue. Plan is to build a system within OpenMV IDE which will allow creating of widget dynamically from the OpenMV Cam.

Undertaker7533967, 2023-07-19:
> machine vision two-offset breakpoint system (Issue #157)

Dear Kwabena :

 

We are glad to hear your plan that is build a system within OpenMV IDE to
allow us to create of widget dynamically from the OpenMV cam

After then we will try this function first~~

Thanks for your plan information ~~ 

 

All the best~~

<https://github.com/openmv/openmv-ide/issues/157#issuecomment-1640957195> ,
3XRBANCNFSM5ZWXB5BQ> .