openmv/openmv-ide #161: Program crashing when trying to flash to a Arduino Nano33 BLE Sense

OKPersson, 2022-07-17:
Same issue as #135

This issue persists on version 2.9.2 on macOS 12.4 Monterey, Thought it was due to the softdevice or bootloader of the nano33 BLE sense, but after a reflash via the arduino ide it's still not working.

_Originally posted by @OKPersson in https://github.com/openmv/openmv-ide/issues/135#issuecomment-1186580735_

kwagyeman, 2022-08-17:
People still see the instant crash... there's definitely something somewhere going wrong. Will search.

Can you post the crash dump log?

OKPersson, 2022-09-12:
I am sorry, I switched to the micropython port instead as it got support for the device around that time. Don't think that I have the logs anymore.

alexrosenblatt, 2022-09-16:
I am seeing it as well - here is the crash log: 

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenMV IDE [4469]
Path:                  /Applications/OpenMV IDE.app/Contents/MacOS/OpenMV IDE
Identifier:            org.openmv.openmvide
Version:               4.0 (4.0.2)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-09-15 21:58:32.6153 -0700
OS Version:            macOS 12.5.1 (21G83)
Report Version:        12
Anonymous UUID:        645987A0-78CE-6FD3-974F-2A153E723F43

Sleep/Wake UUID:       606D9248-4FF0-428A-8ABF-DFF40B0EA57C

Time Awake Since Boot: 10000 seconds
Time Since Wake:       3822 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000c00004e7dd00 -> 0x0000400004e7dd00 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x0000c00004e7dd00
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [4469]

VM Region Info: 0x400004e7dd00 is not in any region.  Bytes after previous region: 69887790144769  Bytes before following region: 35184289784576
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   QtCore                        	       0x10eeaddd4 0x10ee0b000 + 667092
1   QtCore                        	       0x10eeacfa9 operator<(QString const&, QString const&) + 57
2   libOpenMV.dylib               	       0x116c9f77c QMap<QString, QPair<int, int> >::remove(QString const&) + 140
3   libOpenMV.dylib               	       0x116c7f969 OpenMV::Internal::OpenMVPlugin::connectClicked(bool, QString, bool, bool, bool) + 15097
4   libOpenMV.dylib               	       0x116cbb2db QtPrivate::QFunctorSlotObject<OpenMV::Internal::OpenMVPlugin::extensionsInitialized()::$_117, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) + 75
5   QtCore                        	       0x10f024ffe QMetaObject::activate(QObject*, int, int, void**) + 2334
6   QtWidgets                     	       0x10d431925 QAction::activate(QAction::ActionEvent) + 309
7   QtWidgets                     	       0x10d53343e 0x10d429000 + 1090622
8   QtWidgets                     	       0x10d5346ff QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 271
9   QtWidgets                     	       0x10d6165df QToolButton::mouseReleaseEvent(QMouseEvent*) + 15
10  QtWidgets                     	       0x10d475d1f QWidget::event(QEvent*) + 463
11  QtWidgets                     	       0x10d616b81 QToolButton::event(QEvent*) + 321
12  QtWidgets                     	       0x10d43af6d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
13  QtWidgets                     	       0x10d43dcfd QApplication::notify(QObject*, QEvent*) + 7005
14  QtCore                        	       0x10eff4cd8 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 168
15  QtWidgets                     	       0x10d43b8d0 QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) + 928
16  QtWidgets                     	       0x10d495da0 0x10d429000 + 445856
17  QtWidgets                     	       0x10d494a35 0x10d429000 + 440885
18  QtWidgets                     	       0x10d43af6d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
19  QtWidgets                     	       0x10d43c30a QApplication::notify(QObject*, QEvent*) + 362
20  QtCore                        	       0x10eff4cd8 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 168
21  QtGui                         	       0x10e209ddb QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 2171
22  QtGui                         	       0x10e1f038b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 171
23  libqcocoa.dylib               	       0x111e98ece 0x111e71000 + 163534
24  libqcocoa.dylib               	       0x111e99791 0x111e71000 + 165777
25  CoreFoundation                	    0x7ff81f43b37a __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
26  CoreFoundation                	    0x7ff81f43b2e2 __CFRunLoopDoSource0 + 180
[... 395 lines cut ...]
    "queue" : "com.apple.main-thread"
  }
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "6112e14f37f5d11121dcd519",
      "factorPackIds" : {
        "SIRI_TEXT_TO_SPEECH" : "630cd2160c149c2b420228ab"
      },
      "deploymentId" : 240000206
    },
    {
      "rolloutId" : "61301e3a61217b3110231469",
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "6216ae152a40e71046e16225"
      },
      "deploymentId" : 240000016
    }
  ],
  "experiments" : [

  ]
}
}

kwagyeman, 2022-09-18:
Thanks!

This line narrows it down significantly:

2 libOpenMV.dylib 0x116c9f77c QMap<QString, QPair<int, int> >::remove(QString const&) + 140

kwagyeman, 2022-09-18:
Found the bug:

Was doing:

i.remove();
eraseMappings.remove(i.key());
eraseAllMappings.remove(i.key());
vidpidMappings.remove(i.key());

Versus:

eraseMappings.remove(i.key());
eraseAllMappings.remove(i.key());
vidpidMappings.remove(i.key());
i.remove();

...

i.remove() causes the iterator to be invalid. But, I was using it after that.

kwagyeman, 2022-09-18:
This will be fixed in the next IDE release.

closed 2022-09-18 by kwagyeman