openmv/openmv-ide #162: No cursor on terminal spawned from IDE

johncblacker, 2022-07-22:
When I start a terminal from the IDE there is no cursor, so it's not easy to keep track of the command you're typing.  I use REPL a lot and rather than execute my script from main.py, I rename the script after uploading to the board to something like omvmain.py.  Then I can either import <myscript.py> or exec(open('myscript.py', 'r').read())  that way I'm not stuck with a non-responsive board when it gets stuck executing main.py...

kwagyeman, 2022-07-23:
You just want the blinking standard box cursor? This is like a one like of code change for me.

johncblacker, 2022-07-24:
Yes, that's it.  I'm surprised nobody has requested this before...

kwagyeman, 2022-07-24:
Most people don't ever find the feature. :| I added it though because I wanted a good serial terminal.

kwagyeman, 2022-10-02:
Done

closed 2022-10-02 by kwagyeman