openmv/openmv-ide #164: silent install doesn't work with openmv ide v2.9.2 and v2.9.7

yjun123, 2022-10-22:
https://github.com/openmv/openmv-ide#instructions-for-running-the-installer-silently 

According to the above link, I cannot install openmv ide v2.9.7 and v2.9.2 silently, but works for v2.9.0.

```
$ ./openmv-ide-2.9.2-x86_64.run -platform minimal --script silent-install.qs
Unknown option: p, l, a, t, f, o, r, m
```
[silent-install.qs](https://pastebin.mozilla.org/YMe58RbX)

kwagyeman, 2022-10-23:
Will address in the next release.

kwagyeman, 2023-05-05:
Hi, this isn't an official feature. Also, there are some things that people mention about this issue in the link.

Anyway, I'll set this as a feature to support when I finish updating the IDE libraries to the latest Qt, Qt Creator, and Qt Installer framework.

kwagyeman, 2023-07-06:
New Qt Installer framework for the IDE about to be released features command line install mode.

closed 2023-07-06 by kwagyeman