openmv/openmv-ide #165: OpenMV IDE Crash On Import TinyDB

pbhuter, 2022-10-28:


kwagyeman, 2023-05-04:
Thanks for finding this. I have duplicated the error. Fixing now.

kwagyeman, 2023-05-04:
Found the issue. It was because my parser assumed that you'd use the module name for the file versus ".". Anyway, there was a piece of code where I'd try to grab form a list and it could be empty expecting the module name which caused the IDE to crash.

This is now fixed. I also fixed an issue with the scanner computing the wrong hash of the module given the order of files being scanned.

Thanks for find this. I don't use this feature too much and it's been broken. It should work well now.

...

As an aside. Issues with the IDE should get fixed quicker now. Trying to go full-time on OpenMV.

closed 2023-05-04 by kwagyeman