openmv/openmv-ide #166: Feature Request: Auto-reconnect after low power mode

CrispyJawa, 2022-12-05:
After using a command such as pyb.stop() or machine.sleep(), having the ability to reconnect to the IDE and continue making outputs to the serial terminal could be useful.

[Original forum post for additional context](https://forums.openmv.io/t/help-with-using-nicla-vision-and-low-power-mode/7741/4)

kwagyeman, 2022-12-05:
I'll add a checkbox option to do that to the menus in the IDE.

Since this is easy it will come soon...

kwagyeman, 2023-05-05:
Hi, the feature is implemented and working. Will come out this weekend.

closed 2023-05-05 by kwagyeman

kwagyeman, 2023-05-07:
This feature works on the 12 Mb/s OpenMV Cams. However, on 480 Mb/s ones the OS doesn't allow the serial port object to be deleted which prevents the IDE from closing. This works after the serial port is forcibly closed by removing it from the PC.

Not sure how to fix this and it's a bug in the OS...