openmv/openmv-ide #168: Arduino Nicla Vision is not detected on a pure Linux system

port513, 2023-05-04:


kwagyeman, 2023-05-04:
Hi, I've only ever tested the IDE on Ubuntu. Other OSes may have issues. It sound like though that the IDE doesn't have serial port access if it can't find the OpenMV Cam. You may want to check your plugdev group and see that you have permission. If Arduino IDE works you should though.

I'm doing an update of the IDE binaries this month to the latest Qt. This may just be an issue with us using an old library code base. Once I don the update this issue may fix itself.

kwagyeman, 2023-05-04:
Sorry, not exactly sure how to help. Other than, this hopefully gets fixed once the IDE library are all updated.

The IDE should work without issues on Ubuntu.

port513, 2023-05-04:
> Sorry, not exactly sure how to help. Other than, this hopefully gets fixed once the IDE library are all updated.
> 
> The IDE should work without issues on Ubuntu.

Then we enter the wating game and hope that it will fix this weird issue ;)
Thanks for a quick response

port513, 2023-05-06:
Some update that might be helpful.
Just fired up a VM of the same dist I use and it works on that system.
Different number of python3 packages though if that might be something that interfere

On the one that don't find the serial port in IDE
ipython3/parrot,parrot-security,now 7.20.0-1+deb11u1 all [installerat,automatiskt]
libpython3-dev/parrot,now 3.9.2-3 amd64 [installerat,automatiskt]
libpython3-stdlib/parrot,now 3.9.2-3 amd64 [installerat,automatiskt]
libpython3.9-dev/parrot,now 3.9.2-1 amd64 [installerat,automatiskt]
libpython3.9-minimal/parrot,now 3.9.2-1 amd64 [installerat,automatiskt]
libpython3.9-stdlib/parrot,now 3.9.2-1 amd64 [installerat,automatiskt]
libpython3.9/parrot,now 3.9.2-1 amd64 [installerat,automatiskt]
python3-acora/parrot,now 2.2-1.3+b1 amd64 [installerat,automatiskt]
python3-adal/parrot,now 1.2.2-1 all [installerat,automatiskt]
python3-adblockparser/parrot,now 0.7-0parrot1 all [installerat,automatiskt]
python3-advancedhttpserver/parrot,now 2.2.0-0parrot1 all [installerat,automatiskt]
python3-aiocmd/parrot,now 0.1.2-0parrot1 all [installerat,automatiskt]
python3-aioconsole/parrot,now 0.3.1-0parrot1 all [installerat,automatiskt]
python3-aiohttp-socks/parrot,now 0.5.3-1 all [installerat,automatiskt]
python3-aiohttp/parrot,now 3.7.4-1parrot1 amd64 [installerat,automatiskt]
python3-aiorpcx/parrot,now 0.18.5-1 all [installerat,automatiskt]
python3-aiowinreg/parrot,now 0.0.3-0parrot1 all [installerat,automatiskt]
python3-ajpy/parrot,now 0.0.4-2.1 all [installerat,automatiskt]
python3-alembic/parrot,now 1.4.3-1 all [installerat,automatiskt]
python3-aniso8601/parrot,now 8.1.1-1 all [installerat,automatiskt]
python3-antlr4/parrot,now 4.9.1-1 all [installerat,automatiskt]
python3-apispec-webframeworks/parrot,now 0.5.2-0kali1 all [installerat,automatiskt]
python3-apispec/parrot,now 3.3.1-0parrot1 all [installerat,automatiskt]
python3-apparmor/parrot,now 2.13.6-10 amd64 [installerat,automatiskt]
python3-applicationinsights/parrot,now 0.11.9-3 all [installerat,automatiskt]
python3-apt/parrot,now 2.2.1+parrot1 amd64 [installerat,automatiskt]
python3-argcomplete/parrot,now 1.8.1-1.5 all [installerat,automatiskt]
python3-arrow/parrot,now 0.17.0-1 all [installerat,automatiskt]
python3-artifacts/parrot,now 20201106-1 all [installerat,automatiskt]
python3-asciitree/parrot,now 0.3.3-2 all [installerat,automatiskt]
python3-asgiref/parrot,now 3.3.1-1 all [installerat,automatiskt]
python3-asn1crypto/parrot,now 1.4.0-1 all [installerat,automatiskt]
python3-async-timeout/parrot,now 3.0.1-1.1 all [installerat,automatiskt]
python3-asysocks/parrot,now 0.0.2-0parrot1 all [installerat,automatiskt]
python3-attr/parrot,now 20.3.0-1 all [installerat,automatiskt]
python3-autobahn/parrot,now 17.10.1+dfsg1-7 all [installerat,automatiskt]
python3-automat/parrot,now 20.2.0-1 all [installerat,automatiskt]
python3-avro/parrot,now 1.10.1+dfsg-1 all [installerat,automatiskt]
python3-azext-devops/parrot,now 0.19.0-1 all [installerat,automatiskt]
python3-azure-cli-core/parrot,now 2.18.0-2 all [installerat,automatiskt]
python3-azure-cli-telemetry/parrot,now 2.18.0-2 all [installerat,automatiskt]
python3-azure-cli/parrot,now 2.18.0-2 all [installerat,automatiskt]
python3-azure-cosmos/parrot,now 3.1.1-3 all [installerat,automatiskt]
python3-azure-datalake-store/parrot,now 0.0.51-1 all [installerat,automatiskt]
python3-azure-functions-devops-build/parrot,now 0.0.22-5 all [installerat,automatiskt]
python3-azure-multiapi-storage/parrot,now 0.5.2-1 all [installerat,automatiskt]
python3-azure/parrot,now 20201208+git-6 all [installerat,automatiskt]
python3-backcall/parrot,now 0.2.0-1 all [installerat,automatiskt]
python3-bcrypt/parrot,now 3.1.7-4 amd64 [installerat,automatiskt]
python3-binwalk/parrot,now 2.3.1+dfsg1-1 all [installerat,automatiskt]
python3-bitstring/parrot,now 3.1.7-2 all [installerat,automatiskt]
python3-bleach/parrot,now 3.2.1-2.1 all [installerat,automatiskt]
python3-blinker/parrot,now 1.4+dfsg1-0.3 all [installerat,automatiskt]
python3-bluepy/parrot,now 1.3.0-0parrot1 amd64 [installerat,automatiskt]
python3-boltons/parrot,now 19.1.0-3 all [installerat,automatiskt]
python3-boto3/parrot,now 1.13.14-1 all [installerat,automatiskt]
python3-botocore/parrot,now 1.20.0+repack-1 all [installerat,automatiskt]
python3-bottle/parrot,parrot-security,now 0.12.19-1+deb11u1 all [installerat,automatiskt]
python3-brotli/parrot,now 1.0.9-2+b2 amd64 [installerat,automatiskt]
python3-bs4/parrot,now 4.9.3-1 all [installerat,automatiskt]
python3-cairo-dev/parrot,now 1.16.2-4 all [installerat,automatiskt]
python3-cairo/parrot,now 1.16.2-4+b2 amd64 [installerat,automatiskt]
python3-cairocffi/parrot,now 1.2.0-1 all [installerat,automatiskt]
python3-capstone/parrot,now 4.0.2-3 amd64 [installerat,automatiskt]
python3-cbor/parrot,now 1.0.0-1.1+b4 amd64 [installerat,automatiskt]
[... 841 lines cut ...]
python3-xlib/parrot,now 0.29-1 all [installerat,automatiskt]
python3-xlsxwriter/parrot,now 1.1.2-0.2 all [installerat,automatiskt]
python3-xmltodict/parrot,now 0.12.0-2 all [installerat,automatiskt]
python3-yaml/parrot,now 5.3.1-5 amd64 [installerat,automatiskt]
python3-yara/parrot,now 4.0.4-1 amd64 [installerat,automatiskt]
python3-yarl/parrot,now 1.6.3-2 amd64 [installerat,automatiskt]
python3-yaswfp/parrot,now 0.9.3-1.1 all [installerat,automatiskt]
python3-zipp/parrot,now 1.0.0-3 all [installerat,automatiskt]
python3-zmq/parrot,now 20.0.0-1+b1 amd64 [installerat,automatiskt]
[... cut ...]
Python2 on the machine that don't work the one that works don't have any python 2.7 installed att all
libpython2-stdlib/parrot,now 2.7.18-3 amd64 [installerat,automatiskt]
libpython2.7-minimal/parrot,now 2.7.18-8 amd64 [installerat,automatiskt]
libpython2.7-stdlib/parrot,now 2.7.18-8 amd64 [installerat,automatiskt]
python2-minimal/parrot,now 2.7.18-3 amd64 [installerat,automatiskt]
python2.7-minimal/parrot,now 2.7.18-8 amd64 [installerat,automatiskt]
python2.7/parrot,now 2.7.18-8 amd64 [installerat]
python2/parrot,now 2.7.18-3 amd64 [installerat,automatiskt]

kwagyeman, 2023-05-08:
Mmm, yeah, but, the IDE doesn't care about python on the local PC.

port513, 2023-05-08:
Interesting since it works on new installations but not my laptop i stallet one year ago.
But I wait for the new IDE then...

kwagyeman, 2023-05-08:
Cool, it will be about a month. We're using Qt 5.9.1 which is over 5 years old... so, it's going to be a nightmare to update Qt. But, gonna get it done.

kwagyeman, 2023-07-13:
@port513 Hi, the new IDE isn't officially released but I have a working version available. Can you confirm this issue is fixed?

https://github.com/openmv/openmv-ide/releases/download/development/openmv-ide-linux-x86_64-4.0.0.run

kwagyeman, 2023-07-18:
Hi, I found the issue I think for this. It's a bug in the QSerialPortInfo library form Qt. Sometimes on linux the library pulls the USB Hub's VID/PID versus the serial port itselfs VID/PID.

I'll have to do... some big work around to fix this.

kwagyeman, 2023-07-18:
See: https://forums.openmv.io/t/openmv-h7-r1-no-openmv-cams-found/8416/11

iabdalkader, 2023-07-18:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/openmv-h7-r1-no-openmv-cams-found/8416/15

kwagyeman, 2023-07-18:
Hi, I've fixed this issue in the latest IDE. It was as mentioned. The QSerialPortInfo class in Qt was returning the USB Hub VID/PID and not the Port VID/PID which will filter by.

closed 2023-07-18 by kwagyeman