openmv/openmv-ide #170: IDE download was blocked by Advanced Protection in Chrome

erichiggins, 2023-05-14:
Trying to download the OpenMV IDE (for MacOS) using the Google Chrome browser and an account with [Advanced Protection](https://landing.google.com/advancedprotection/) enabled, the download is blocked.

The "Learn More" link points to the following URL:
https://support.google.com/accounts/answer/7539956?hl=en&visit_id=638196929201939466-536696130&p=safe-browsing&rd=1#troubleshoot_safe_browsing&zippy=%2Cwhy-did-chrome-stop-my-download

### Screenshot
[image]

kwagyeman, 2023-05-14:
Hi, you just gotta say it's okay. The whole binary is signed by apple's gate keeper software. However, I guess chrome doesn't know this.

Can you send the file to be scanned?

(As an aside, the thing is submitted to apple to be verified. This is done on each binary in the APP, then the APP itself, then the DMG made from the APP. Now it looks like Chrome want's in on having to verify things too).

erichiggins, 2023-05-16:
@kwagyeman The file that was flagged was just the MacOS download from the OpenMV website. Chrome did not give me the option to send it to be scanned in this case.

There may be other reasons why this is being flagged than it having been approved by Apple, though Google is unlikely to list them since it may inform bad actors on how to avoid those checks.

kwagyeman, 2023-05-16:
I don't know if Google coordinates with Apple. The IDE is signed by gatekeeper however.

I'll leave this open to see if there's a program to submit the IDE to Google for scanning too.

kwagyeman, 2023-07-13:
Hi, I just looked into this more. There's no developer option to send the file in to be scanned.

The only option is for you to manually send the file in when you are downloading. If this option is not presented then it may have been hidden given your chrome settings.

Anyway, be assured that we do sign our program and send it to apple's gate keeper software which would not let you run it without that signature. Similarly, for windows we do the same time and digitally sign the program. Finally, on linux our IDE does not require admin privileges to run.

closed 2023-07-13 by kwagyeman