openmv/openmv-ide #178: Multilingual not available in new version.

kidswong999, 2023-09-20:
Since the IDE was updated to 4.0.1, multilinguals are no longer available. 
Can this feature be fixed?

kwagyeman, 2023-09-20:
What do you mean? The IDE has been translated into 32 languages.

kidswong999, 2023-09-21:
Language selection in preference does not work.
I chose Chinese and restarted the software, but the interface is still in English.

kwagyeman, 2023-09-21:
Is this on a Mac? Which OS?

But, yes, can fix this soon.

kidswong999, 2023-09-22:
I tested Windows10 and MacOS, and neither worked.

kwagyeman, 2023-09-22:
Windows should have works. Okay, will determine what is broken and fix it ASAP.

kwagyeman, 2023-09-25:
Please let me know if the issue is fixed here: https://github.com/openmv/openmv-ide/releases/tag/development

kwagyeman, 2023-09-26:
This seems to work now.

kidswong999, 2023-09-29:
Yes, it works now.