openmv/openmv-ide #193: Ubuntu 22.04: Could not load the Qt platform plugin "xcb" in "" even though it was found.

xue-singtown, 2024-01-12:
I can't launch OpenMV IDE in Ubuntu 22.04.

I have installed libxcb-xinerama0.

Version: `openmv-ide-linux-x86_64-4.1.5.tar.gz`

Command:
`~/Downloads/openmv-ide/bin$ ./openmvide`

```
Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, vkkhrdisplay, minimalegl, wayland, xcb, minimal, offscreen, eglfs, wayland-egl, vnc.
```

Thanks.

dewhisna, 2024-01-12:
According to: https://stackoverflow.com/questions/76802977/ubuntu-22-04-qt-6-5-qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb, installing "libxcb-cursor0" might fix the problem.  And if not, their debug procedure might help track it down.

kwagyeman, 2024-01-12:
Did you run the setup script? It install a bunch of stuff you need.

xue-singtown, 2024-01-14:
I installed `libxcb-cursor0`, it works now.

closed 2024-01-15 by kwagyeman