openmv/openmv-ide #201: Bookworm installation: python 3.11+, libicuuc.so.67, and qt 5.15.0

matiasandina, 2024-06-05:
I'm trying to install in the newest Raspberry Pi Os (bookworm). I downloaded the Bullseye installer (thinking it might not work, but just to give it the chance).

Bookworm ships with python 3.11, which yells at you if you're doing anything outside a venv.

Ide installation fails due to due to the `sudo pip install pyusb` line where

```
error: externally-managed-environment

x This environment is externally managed
--> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

Is there a reason why using pip for this particular install? 
I replaced it with this line:

```
sudo apt install python3-usb 
```

The install goes through. I get the icon and everything, but 
```
~/Downloads/openmv-ide $ ./bin/openmvide
./bin/openmvide: error while loading shared libraries: libicuuc.so.67: cannot open shared object file: No such file or directory
```

I have a newer version of libicuuc.so, so maybe that's getting into trouble? I would appreciate some help here

```
~/Downloads/openmv-ide $ sudo find / -name 'libicu*'
/usr/lib/aarch64-linux-gnu/libicudata.so.72
/usr/lib/aarch64-linux-gnu/libicuuc.so.72
```

I solved this by performing some symlinks.

I went on to run the ide, but I got stuck into a qt5 incompatibility error. The install that I have is `5.15.8`, not sure how incompatible it is.

```
pi@raspberrypi:~/Downloads/openmv-ide $ ./bin/openmvide
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqxcb.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqwayland-generic.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqlinuxfb.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqeglfs.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqwayland-xcomposite-egl.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqwayland-xcomposite-glx.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqwayland-egl.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqminimalegl.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqvnc.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqminimal.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
In /usr/lib/aarch64-linux-gnu/qt5/plugins/platforms/libqoffscreen.so:
  Plugin uses incompatible Qt library (5.15.0) [release]
Could not find the Qt platform plugin "wayland" in "/usr/lib/aarch64-linux-gnu/qt5/plugins/platforms"
Could not load the Qt platform plugin "xcb" in "/usr/lib/aarch64-linux-gnu/qt5/plugins/platforms" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, linuxfb, eglfs, minimalegl, vnc, minimal, offscreen.

Aborted
```

Any help appreciated!

kwagyeman, 2024-06-05:
Hi, the basic problem is that the build script for the IDE doesn't include all the libraries it's built with in the installer. This is the case for linux/mac/windows. However, there was not a lot of technical support for building for the Pi. I barely managed to make it all work.

The way to fix it is just to include those missing libraries inside of the IDE install dir with the right name.

As for the Qt incompatibility error, it shouldn't be trying to load the Qt5 libraries. The IDE is built on Qt6. It should have never searched those library paths...

As for pip, that's just installed to make some fallback scripts work when flashing via DFU. It's not super important. But, I'll try to fix that in the installer build.

matiasandina, 2024-06-05:
I was digging about the error and found a related [SO question](https://stackoverflow.com/questions/76802977/ubuntu-22-04-qt-6-5-qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb)

So I just tried to do:

```
sudo apt-get install libxcb-cursor0
```

This resulted in the first launch image loading.

But the GUI it crashed with:

```
~/Downloads/openmv-ide $ ./bin/openmvide
Could not find the Qt platform plugin "wayland" in ""
./bin/openmvide: symbol lookup error: /home/pi/Downloads/openmv-ide/bin/../lib/Qt/lib/libQt6Core5Compat.so.6: undefined symbol: ucnv_getDefaultName_67
```
I feel it's almost there, and it's about work, Qt5 stuff is no longer printed.

kwagyeman, 2024-06-05:
Hi, do ldd on: /home/pi/Downloads/openmv-ide/bin/../lib/Qt/lib/libQt6Core5Compat.so.6

> Could not find the Qt platform plugin "wayland" in ""

These errors are generally that something didn't load. Unfortunately, they tell you almost nothing since it could be any reason why it didn't load.

kwagyeman, 2024-06-05:
At least now it's loading libraries from the install folder.

matiasandina, 2024-06-05:
This is what comes out

```
ldd /home/pi/Downloads/openmv-ide/bin/../lib/Qt/lib/libQt6Core5Compat.so.6
linux-vdso.so.1 (0x0000007fa3115000)
libicuuc.so.67 => /lib/aarch64-linux-gnu/libicuuc.so.67 (0x0000007fa2e40000)
libQt6Core.so.6 =>
/home/pi/Downloads/openmv-ide/bin/../lib/Qt/lib/libQt6Core.so.6
(0x0000007fa2850000)
libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007fa2630000)
libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007fa2480000)
/lib/ld-linux-aarch64.so.1 (0x0000007fa30d8000)
libicudata.so.72 => /lib/aarch64-linux-gnu/libicudata.so.72
(0x0000007fa0690000)
libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007fa05f0000)
libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007fa05b0000)
libicui18n.so.67 => /lib/aarch64-linux-gnu/libicui18n.so.67
(0x0000007fa0270000)
libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007fa0240000)
libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0
(0x0000007fa00e0000)
libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007fa00a0000)
libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0
(0x0000007fa0070000)
libpcre2-16.so.0 => /lib/aarch64-linux-gnu/libpcre2-16.so.0
(0x0000007f9ffd0000)
libpcre2-8.so.0 => /lib/aarch64-linux-gnu/libpcre2-8.so.0
(0x0000007f9ff20000)
```

kwagyeman, 2024-06-05:
This appears to be the a similar issue with the libicui stuff. ucnv_getDefaultName is a function in libcui. So, I think what's happening is that the function the library needs is missing from your updated version...

kwagyeman, 2024-06-05:
Yeah, if they didn't include that version of it, then in the .72 lib you have, then, you'll need an older copy of that library.

kwagyeman, 2024-06-05:
Starting next month I will be working on updating the IDE again and will fix these issues by creating a new build root using bookworm along with including these libraries in the release.

matiasandina, 2024-06-05:
Ok appreciated! I am a bit hesitant to roll back these libraries to older versions, so I guess maybe I should wait on it ? What do you suggest?

kwagyeman, 2024-06-05:
Dunno. I'd try to get a copy of the libs and put them into the IDE install folder.

matiasandina, 2024-06-05:
So...I downloaded the files I *think* (really not sure) are the ones I need from

```
wget http://ftp.us.debian.org/debian/pool/main/i/icu/libicu67_67.1-7_armhf.deb
mkdir icu67
dpkg-deb -x libicu67_67.1-7_armhf.deb icu67
```

Then I did

```
cp -rv ~/Downloads/openmv-ide/icu67/usr/lib/arm-linux-gnueabihf/* ~/Downloads/openmv-ide/lib/Qt/lib/
```

Then tried to run it, but same error.

```
~/Downloads/openmv-ide $ ./bin/openmvide
Could not find the Qt platform plugin "wayland" in ""
./bin/openmvide: symbol lookup error: /home/pi/Downloads/openmv-ide/lib/Qt/lib/libQt6Core5Compat.so.6: undefined symbol: ucnv_getDefaultName_67
```

```
~/Downloads/openmv-ide $ ls /home/pi/Downloads/openmv-ide/bin/../lib/Qt/lib/
libicudata.so.67    libQt6Concurrent.so.6                  libQt6Network.so.6           libQt6Svg.so.6
libicudata.so.67.1  libQt6Concurrent.so.6.5.1              libQt6Network.so.6.5.1       libQt6Svg.so.6.5.1
libicui18n.so.67    libQt6Core5Compat.so.6                 libQt6OpenGL.so.6            libQt6SvgWidgets.so.6
libicui18n.so.67.1  libQt6Core5Compat.so.6.5.1             libQt6OpenGL.so.6.5.1        libQt6SvgWidgets.so.6.5.1
libicuio.so.67      libQt6Core.so.6                        libQt6PrintSupport.so.6      libQt6Test.so.6
libicuio.so.67.1    libQt6Core.so.6.5.1                    libQt6PrintSupport.so.6.5.1  libQt6Test.so.6.5.1
libicutest.so.67    libQt6DBus.so.6                        libQt6Qml.so.6               libQt6Widgets.so.6
libicutest.so.67.1  libQt6DBus.so.6.5.1                    libQt6Qml.so.6.5.1           libQt6Widgets.so.6.5.1
libicutu.so.67      libQt6EglFSDeviceIntegration.so.6      libQt6SerialPort.so.6        libQt6XcbQpa.so.6
libicutu.so.67.1    libQt6EglFSDeviceIntegration.so.6.5.1  libQt6SerialPort.so.6.5.1    libQt6XcbQpa.so.6.5.1
libicuuc.so.67      libQt6Gui.so.6                         libQt6Sql.so.6               libQt6Xml.so.6
libicuuc.so.67.1    libQt6Gui.so.6.5.1                     libQt6Sql.so.6.5.1           libQt6Xml.so.6.5.1

```

```
~/Downloads/openmv-ide $ ldd /home/pi/Downloads/openmv-ide/bin/../lib/Qt/lib/libQt6Core5Compat.so.6
	linux-vdso.so.1 (0x0000007f95722000)
	libicuuc.so.67 => /lib/aarch64-linux-gnu/libicuuc.so.67 (0x0000007f95450000)
	libQt6Core.so.6 => /home/pi/Downloads/openmv-ide/bin/../lib/Qt/lib/libQt6Core.so.6 (0x0000007f94e60000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f94c40000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f94a90000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f956e5000)
	libicudata.so.72 => /lib/aarch64-linux-gnu/libicudata.so.72 (0x0000007f92ca0000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f92c00000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f92bc0000)
	libicui18n.so.67 => /lib/aarch64-linux-gnu/libicui18n.so.67 (0x0000007f92880000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f92850000)
	libglib-2.0.so.0 => /lib/aarch64-linux-gnu/libglib-2.0.so.0 (0x0000007f926f0000)
	libz.so.1 => /lib/aarch64-linux-gnu/libz.so.1 (0x0000007f926b0000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f92680000)
	libpcre2-16.so.0 => /lib/aarch64-linux-gnu/libpcre2-16.so.0 (0x0000007f925e0000)
	libpcre2-8.so.0 => /lib/aarch64-linux-gnu/libpcre2-8.so.0 (0x0000007f92530000)
```

I'm feeling a bit stuck

kwagyeman, 2024-06-05:
Hi, the sysroot the IDE was built with is here: https://github.com/openmv/qt-raspi/blob/main/rpi-sysroot.tar.gz

You should be able to find the right version of the libraries in there.

kwagyeman, 2024-06-05:
I'll be updating the https://github.com/openmv/qt-raspi to the book worm release in a few months. That said, I may wish to wait until they do the next release so that I'm in sync.

matiasandina, 2024-06-05:
Oh well...it turned out that I had done wget on the wrong file...the correct one is

```
wget http://ftp.us.debian.org/debian/pool/main/i/icu/libicu67_67.1-7_arm64.deb
```
exporting the proper LD_LIBRARY_PATH finally opens the IDE!!!

```
#!/bin/bash
# Setting up environment variables
export LD_LIBRARY_PATH=/home/pi/Downloads/openmv-ide/lib/Qt/lib:$LD_LIBRARY_PATH
# Add any other environment setups here if necessary

# Starting the application
/home/pi/Downloads/openmv-ide/bin/openmvide
```

I made a small patch on the launcher so that I can have this only for the time being. It's far from ideal but I kinda needed to do experiments right now and my previous install in Raspbian Buster is way way out of date!

kwagyeman, 2024-06-05:
Can you just post your steps here on this issue below I close it? For others who are suffering until I have time to update the IDE?

matiasandina, 2024-06-05:
I cannot say for sure, I might be missing some step but:

1. download the bullseye version and extract it to desired location (in my case `~/Downloads/openmv-ide`)
1. change the line in `setup.sh` from `sudo pip install pyusb` with `sudo apt install python3-usb`
1. perform install as `README.md` informs.
1. download the libraries from `wget http://ftp.us.debian.org/debian/pool/main/i/icu/libicu67_67.1-7_arm64.deb` (if having 32 bit system use the other link I provided)
1. extract them onto the desired place. In my case
```
mkdir icu67
dpkg-deb -x libicu67_67.1-7_armhf.deb icu67
```
1. copy the files over
```
cp -rv ~/Downloads/openmv-ide/icu67/usr/lib/arm-linux-gnueabihf/* ~/Downloads/openmv-ide/lib/Qt/lib/
```
1. To point to the proper libraries, it's easier and more reproducible from a bash script. Inside the openmv installation folder do:
```
nano ~/Downloads/openmv-ide/start-openmvide.sh
```

Copy this script and save it to the extraction folder.

```
#!/bin/bash
# Setting up environment variables
export LD_LIBRARY_PATH=/home/pi/Downloads/openmv-ide/lib/Qt/lib:$LD_LIBRARY_PATH
# Add any other environment setups here if necessary

# Starting the application
/home/pi/Downloads/openmv-ide/bin/openmvide
```
1. make the script executable with `chmod +x ~/Downloads/openmv-ide/start-openmvide.sh`
1. Then you can use it from terminal. If you want the desktop icon to work, edit the `Exec`, for example `Exec=/home/pi/Downloads/openmv-ide/start-openmvide.sh`

closed 2024-06-05 by matiasandina

iabdalkader, 2024-06-16:
This issue has been mentioned on **OpenMV Forums**. There might be relevant details there:

https://forums.openmv.io/t/how-can-i-download-openmv-ide-on-rasberry-pi-5/9702/2

rishabsingh3003, 2024-10-09:
This worked for me, thanks! For everyone else, don't forget to first do sudo apt-get install libxcb-cursor0

KurtE, 2024-10-19:
I just tried it today.
It is now coming up: after installed the libxcb-cursor0 mentioned by @rishabsingh3003 

It was able to download the firmware to the GIGA board, and then with the display.py sketch it was
able to display images from OV5640 camera.

I do still get some messages in the terminal window:
```
kurte@RPI5:~/Downloads/openmv-ide $ ./start-openmvide.sh 
Could not find the Qt platform plugin "wayland" in ""
Incompatible version of OpenSSL (built with OpenSSL 1.x, runtime version is >= 3.x)
Corrupt JPEG data: premature end of data segment
```
Not sure if any of these are issues.  I have received a few of the Corrupt messages.

Thanks everyone

kwagyeman, 2024-10-19:
The corrupt messages are because the jpeg file being transferred was corrupt.