openmv/openmv-ide #257: Add a Multi-User install

AlexGrouard, 2024-11-29:
Is it possible to not install all firmware directly on the user Roaming folder ?

I'm trying to deploy it to schools, with each student having their own profile. 
So, can you move the content of the OpenMV roaming folder to programData ?
This way all profiles can use the same folder without cluttering each user session.

Thanks

kwagyeman, 2024-11-29:
Sure, I can look into this. As long as it's writable by the application it's possible to make it a multi-user install.

kwagyeman, 2025-04-27:
@AlexGrouard - I just finished updating major pieces of the IDE and I'll now deal with minor bugs and feature improvements.

Why is this feature needed exactly? Is it because the number of students is very large, causing the copies of the user folder to eat up a huge amount of space?

AlexGrouard, 2025-04-28:
Hello, 
Mostly, yes we can have up to a hundred different accounts on the same machines, thus eating more space than necessary.
Also, we deploy our PC mostly via an automated system, and having the possibility to install it globally will make it largely easier to manage.
Thanks for taking the time to working on this.

kwagyeman, 2025-04-30:
This is done. I've moved all resources to a shared folder. Need to test how this will work under all OSes. But, it should prevent shared resources from being copied.

closed 2025-04-30 by kwagyeman

AlexGrouard, 2025-04-30:
Thanks,
I will try it on Ubuntu 22.04/24.04 and Win10/11.

kwagyeman, 2025-05-04:
@AlexGrouard - I can only enable this for Windows. For Linux, there's no way an app can write to anything except the user folder without admin privileges.

The only way to get around this is to create a shared folder when using root so that the app can write to it.