openmv/openmv-ide #271: OpenMV IDE 4.5.0 Crash on macOS 15.4.1 (Apple Silicon)

Nachobr, 2025-04-30:
macOS version (15.4.1), and hardware (e.g., M1).

Steps: “Installed OpenMV IDE 4.5.0, launched app, crashed immediately.”

Haz clic en Reiniciar para volver a abrir la aplicación. Haz clic en Reporte para ver información más detallada y enviar un reporte a Apple.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenMV IDE [13900]
Path:                  /Applications/OpenMV IDE.app/Contents/MacOS/OpenMV IDE
Identifier:            io.openmv.openmvide
Version:               4.5.0 (4.5.0)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-30 19:54:37.8193 -0300
OS Version:            macOS 15.4.1 (24E263)
Report Version:        12
Anonymous UUID:        4211EDD8-592E-8421-34E7-9299745654B5

Sleep/Wake UUID:       98EF56EF-6C03-4411-8549-62EE85977A67

Time Awake Since Boot: 350000 seconds
Time Since Wake:       2769 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xacbd3346e2c38831 -> 0xffffb346e2c38831 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0xacbd3346e2c38831

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [13900]

VM Region Info: 0xffffb346e2c38831 is not in any region.  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      UNUSED SPACE AT END

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libAggregation.4.5.0.dylib    	       0x10af2e7b0 QHash<QObject*, Aggregation::Aggregate*>::remove(QObject* const&) + 128
1   libAggregation.4.5.0.dylib    	       0x10af2e623 Aggregation::Aggregate::deleteSelf(QObject*) + 83
2   QtCore                        	       0x1101a4495 0x1100e7000 + 775317
3   QtCore                        	       0x11019b921 QObject::destroyed(QObject*) + 65
4   QtWidgets                     	       0x10cf5d985 QWidget::~QWidget() + 1173
5   libCore.dylib                 	       0x1202f403e Core::OutputWindow::~OutputWindow() + 14
6   libCore.dylib                 	       0x1202c5316 Core::Internal::MessageOutputWindow::~MessageOutputWindow() + 38
7   libsystem_c.dylib             	    0x7ff81426eea2 __cxa_finalize_ranges + 408
8   libsystem_c.dylib             	    0x7ff81426ecbd exit + 35
9   libOpenMV.dylib               	       0x123785905 OpenMV::Internal::OpenMVPlugin::initialize(QList<QString> const&, QString*) + 12021
10  libExtensionSystem.4.5.0.dylib	       0x10afe54f8 ExtensionSystem::CppPluginSpec::initializePlugin() + 136
11  libExtensionSystem.4.5.0.dylib	       0x10afc75ed ExtensionSystem::Internal::PluginManagerPrivate::loadPlugin(ExtensionSystem::PluginSpec*, ExtensionSystem::PluginSpec::State) + 1101
12  libExtensionSystem.4.5.0.dylib	       0x10afbbf31 ExtensionSystem::Internal::PluginManagerPrivate::loadPlugins() + 209
13  OpenMV IDE                    	       0x10291550f 0x102907000 + 58639
14  dyld                          	       0x202d19530 start + 3056

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7ffd6a0e4 0x7ff7ffd66000 + 16612

Thread 2:
0   runtime                       	    0x7ff7ffd889b0 0x7ff7ffd66000 + 141744

Thread 3:
0   runtime                       	    0x7ff7ffd889b0 0x7ff7ffd66000 + 141744
[... 129 lines cut ...]
9   libsystem_pthread.dylib       	    0x7ff8143a4df1 _pthread_start + 99
10  libsystem_pthread.dylib       	    0x7ff8143a0857 thread_start + 15

Thread 16:
0   runtime                       	    0x7ff7ffd889b0 0x7ff7ffd66000 + 141744

Thread 17:
0   runtime                       	    0x7ff7ffd889b0 0x7ff7ffd66000 + 141744

Thread 18:
0   runtime                       	    0x7ff7ffd889b0 0x7ff7ffd66000 + 141744

Thread 19:
0   runtime                       	    0x7ff7ffd889b0 0x7ff7ffd66000 + 141744

Thread 20:
0   runtime                       	    0x7ff7ffd889b0 0x7ff7ffd66000 + 141744

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x000000010af350f0  rcx: 0x0000600001e5df40  rdx: 0xacbd3346e2c38820
  rdi: 0x00006000021423d0  rsi: 0x0000000000000000  rbp: 0x0000000309734640  rsp: 0x0000000309734630
   r8: 0x0000000000000011   r9: 0x00000000000001ff  r10: 0x000000007ea00000  r11: 0x000000000000015b
  r12: 0x0000000202dd8200  r13: 0x00006000021424f0  r14: 0x998bbbcd3b587911  r15: 0x0000000202dd8200
  rip: 0x000000010af2e7b0  rfl: 0x0000000000000206
 tmp0: 0x0000000000000000 tmp1: 0x000000010af2e5fc tmp2: 0x000000010af2e623

kwagyeman, 2025-04-30:
It looks like the IDE called exit(-1) because of a failure.

This is happening most likely because a dir it needs to write to is not writable. It should have generated a warning message though via a prompt about something.

Please make sure ~/.config/OpenMV is writable by the IDE.

Nachobr, 2025-05-01:
Perfect, i trained grok with your answer, and from the answer from one of the users on the forum, and now it works, the solution was to create the directory .config/OpenMV, and then give permissions. Thanks!

closed 2025-05-01 by kwagyeman